soffchen/NanoPi-R2S #10: 想请问下R2S的固件。有没有希望走openwrt 的方式直接升级。

blakejia, 2020-03-02:


eluotao, 2020-03-07:
同问.

iotsys, 2020-03-13:
可以参考这个实现：[大神怎么直接在路由器里面直接更新固件 #47](https://github.com/klever1988/nanopi-openwrt/issues/47)

iotsys, 2020-03-13:
可以暂时先参考下这个，不过我是把*.img.zip转成*.img.gz了：[大神怎么直接在路由器里面直接更新固件 #47](https://github.com/klever1988/nanopi-openwrt/issues/47)

klever1988, 2020-03-13:
> 可以暂时先参考下这个，不过我是把*.img.zip转成*.img.gz了：[大神怎么直接在路由器里面直接更新固件 #47](https://github.com/klever1988/nanopi-openwrt/issues/47)

应该是解压出zip包里面的img.gz文件，把这个文件命名成rom.img.gz，然后再执行那三条命令

iotsys, 2020-03-14:
> > 可以暂时先参考下这个，不过我是把*.img.zip转成*.img.gz了：[大神怎么直接在路由器里面直接更新固件 #47](https://github.com/klever1988/nanopi-openwrt/issues/47)
> 
> 应该是解压出zip包里面的img.gz文件，把这个文件命名成rom.img.gz，然后再执行那三条命令

是的，谢谢大佬指点。主要是这边编译生成的.img.zip格式，所以我解压后转了一下格式。不知道unzip能不能这样操作，我回头试试

klever1988, 2020-03-14:
可以unzip的

azgh, 2020-04-22:
@klever1988 R2S minimal 发现两个奇怪现象，没法反馈，不好意思在此打扰了：

1. 概况界面，接口eth0变成了100Mb/s全双工，原本应该是1000Mb/s；
2. 概况界面，网络IPv4 WAN状态的地址，显示的不是真实地址。

请核实，谢谢。

soffchen, 2020-05-08:
最新版本提供 luci 的方式来更新固件

closed 2020-05-08 by soffchen