soffchen/NanoPi-R2S #12: 有没有计划加个 sfe

fanck0605, 2020-03-03:
这机子的nat的不咋行啊，有没有计划加个  flowoffload(这玩也好像19.07自带了) 或者 sfe 这种nat加速组件啊

soffchen, 2020-03-03:
flowoffload 内核自带了，但没有整合到 iptables 的样子，有在看。

pwrliang, 2020-03-03:
@soffchen Hi，请问您在使用的时候有没有发现在防火墙添加端口转发规则后（可能需要多试几次），点击“保存&应用”会导致断网，只能拔电源重启。貌似和fullcone-nat有关，在防火墙中把“启用FullCone-NAT”关掉，就很少会遇到这种情况了。

fanck0605, 2020-03-03:
大佬加油搞啊。那个配置文件最好是用 menuconfig 生成的，毕竟手动改总有失误

soffchen, 2020-03-12:
最新支持了 Flow Offloading

closed 2020-03-12 by soffchen

fanck0605, 2020-03-28:
在R2S PPPoE 拨号的情况下，Flow Offloading 可能会导致 IPv4 彻底瘫痪，甚至无法使用 IPv4 访问路由器，IPv6 不受影响。

soffchen, 2020-03-28:
对的，所以默认关闭了，实测不开 Flow Offloading  NAT 性能也不差，先这样吧。

fanck0605, 2020-03-28:
🤣老版本的固件，刚刷完固件就进不了后台，有点懵，应该是 firewall 崩了。
隔壁 @klever1988 的固件也有这个问题，他用了个守护脚本尝试解决，不知道效果如何。

soffchen, 2020-03-28:
老版本的固件？指的是哪个？前天发布的我测试 ok 的

fanck0605, 2020-03-29:
新版本仍然有问题，应该是内核补丁导致的，你和 @klever1988 的版本都有这个问题

复现方式：[请问使用Lienol版固件的R2S用户，你们会出现不明原因的死机情况吗？](https://github.com/klever1988/nanopi-openwrt/issues/94)

soffchen, 2020-03-29:
我来把内核补丁都去掉看看

fanck0605, 2020-03-29:
[这里](https://github.com/friendlyarm/friendlywrt/commit/f092ca098e80c667b10cdd1dba328506a2673c1d#commitcomment-37535668)也有提到无法使用ipv4的情况，这可能不是内核补丁的问题。