soffchen/NanoPi-R2S #27: 请问能集成smartdns和frpc吗？

gzelvis, 2020-05-07:
请问你的固件可以加入smartdns和frpc吗？实在感谢

soffchen, 2020-05-08:
frpc 本来就有哦

smartdns 加进来了，一会儿更新。

gzelvis, 2020-05-08:
嗯嗯，还有clash昨天也更新到0.20版本了，兄弟一块update憋，辛苦了

gzelvis, 2020-05-08:
忘了说，能否集成 tcp_htcp? 其实作为客户端（国内机），bbr意义不大，但htcp能很大程度提升响应 

另外，默认的sysctl.conf，我这里给你一个最优化方案参考。你看设一下固件里的默认？

fs.file-max = 104400

net.core.rmem_default = 31457280
net.core.wmem_default = 31457280
net.core.rmem_max = 67108864
net.core.wmem_max = 67108864
net.core.netdev_max_backlog = 250000
net.core.somaxconn = 65535
net.core.optmem_max = 25165824

net.ipv4.tcp_syncookies = 1
net.ipv4.tcp_synack_retries = 2
net.ipv4.tcp_rfc1337 = 1
net.ipv4.tcp_tw_reuse = 1
net.ipv4.tcp_tw_recycle = 0
net.ipv4.tcp_fin_timeout = 30
net.ipv4.tcp_keepalive_time = 1200
net.ipv4.ip_local_port_range = 10000 65535

net.ipv4.tcp_max_syn_backlog = 25000
net.ipv4.tcp_max_tw_buckets = 1440000
net.ipv4.tcp_fastopen = 3
net.ipv4.tcp_mem = 65536 131072 262144 
net.ipv4.udp_mem = 65536 131072 262144 
net.ipv4.tcp_rmem = 4096 87380 67108864
net.ipv4.tcp_wmem = 4096 87380 67108864
net.ipv4.tcp_mtu_probing = 1
net.ipv4.ip_forward = 1
net.ipv4.tcp_congestion_control = htcp

gzelvis, 2020-05-08:
这里有个问题，上面的配置是我x86 软路由的配置，R2s的话鉴于性能，估计 net.core.somaxconn 这个只能设置为16384或者25000。 net.core.netdev_max_backlog 缩小为 25000。或者你也测试看看？因为我的R2s估计要下周一才能收到了

tefiszx, 2020-05-08:
> 嗯嗯，还有clash昨天也更新到0.20版本了，兄弟一块update憋，辛苦了

clash可以在luci界面用按钮更新吧

gzelvis, 2020-05-08:
> > 嗯嗯，还有clash昨天也更新到0.20版本了，兄弟一块update憋，辛苦了
> 
> clash可以在luci界面用按钮更新吧

抱歉我的R2s还没收到。但之前在N1上看的那个Openclash繁多复杂的功能和设置项，还有锁死的端口，我这个x86软路由用了一年多clash的人也被搞蒙圈了。太华而不实了

soffchen, 2020-05-08:
clash 我集成更新了，回头到 actions 里面去下，就不单独 release 了

集成主要是为了更新的时候可以正常、快速 :)

就是鸡蛋跟鸡的关系。

soffchen, 2020-05-08:
> 忘了说，能否集成 tcp_htcp? 其实作为客户端（国内机），bbr意义不大，但htcp能很大程度提升响应
>
> 另外，默认的sysctl.conf，我这里给你一个最优化方案参考。你看设一下固件里的默认？
>
> fs.file-max = 104400
>
> net.core.rmem_default = 31457280
> net.core.wmem_default = 31457280 [...]

等你 r2s 到了以后实测一下再说 :)

gzelvis, 2020-05-08:
> 等你 r2s 到了以后实测一下再说 :)

好的，就是兄弟知道怎么在openwrt里添加 tcp_htcp的modprobe吗？linux很容易，但openwrt貌似是编译的时候就要想办法在内核集成进去？

soffchen, 2020-05-09:
这种 tcp 调度算法优化，不建议在家里路由器上使用，就好比 bbr 打开之后，会影响 wifi 性能。

closed 2020-05-15 by soffchen