soffchen/NanoPi-R2S #34: FRP 服务未运行

Omoinemie, 2020-06-06:
frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致
这个是什么原因，我设置肯定是对的

soffchen, 2020-06-06:
请检查服务器端版本与客户端版本是否一致，frp 对两端版本要求比较高。

Omoinemie, 2020-06-06:
脚本用的是什么版本？不应该是最新版本吗

soffchen, 2020-06-06:
0.33.0 最近版本

与上游 `https://github.com/coolsnowwolf/lede/tree/master/package/lean/frp` 一致

Omoinemie, 2020-06-06:
服务器版本也是0.33.0

Omoinemie, 2020-06-06:
程序权限的问题？

soffchen, 2020-06-06:
`https://github.com/coolsnowwolf/lede/issues/738`

上游也有人反映，也有人提到权限问题，我这边测试是好的 @@

orestesza, 2020-07-02:
遇到相同问题。

iouAkira, 2020-07-23:
**遇到了一样的问题**
[image]

Joachimzhang, 2020-10-17:
遇到了相同的问题

slb116763814, 2020-12-15:
> 请检查服务器端版本与客户端版本是否一致，frp 对两端版本要求比较高。

 有机会集成 RTL8192EU RTL8188EU么？因为正好有这两网卡，不需要去买了。哈哈。

soffchen, 2021-01-24:
测试了一下，应该是由于 frpc 的 bin 被 upx 压缩之后出的问题，新版去掉 upx 压缩，看看哦。

soffchen, 2021-01-25:
新版已经搞定

closed 2021-01-25 by soffchen

chenmo-tssz, 2021-12-08:
我在openwrt上已经成功穿透了，不过今天怎么发现提示 “frp服务启动失败，请检查服务端 “TCP多路复用(tcp_mux)”设置，确保与客户端完全一致” 这个错误

q943613777, 2021-12-08:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。