soffchen/NanoPi-R2S #50: 有机会集成 passwall么

oimken, 2020-10-05:
不知道有没有机会集成 passwall， 自己试着做下面的修改，但是编译失败了

https://github.com/soffchen/NanoPi-R2S/blob/7f428ab2922803becf94dd1139c1480753d6abac/.github/workflows/NanoPi-R2S%20RK3328%20OpenWrt%2019.07%20Build.yml#L58 后面加入
`echo 'src-git songchenwen https://github.com/songchenwen/openwrt-package.git' >> feeds.conf.default`

在`r2s-rk3328-config` 也加入了对应的passwall配置，但还是失败了。

tefiszx, 2020-10-06:
没有更新feeds？

oimken, 2020-10-06:
我用的那个是 songchenwen 的r2s 中使用的，现在看来可能真的因为时间太久了，我试试换feed看看行不行，非常感谢你的帮助！

oimken, 2020-10-06:
> 有什么错误提示没有？我感觉你添加的软件源也不对。要集成passwall试试看添加这个feed
> https://github.com/Lienol/openwrt-package.git
@tefiszx 还是失败了

修改 `NanoPi-R2S RK3328 OpenWrt 19.07 Build.yml`
https://github.com/oimken/NanoPi-R2S/blob/4c48929e461b28b0aba21784855e3e23bca38794/.github/workflows/NanoPi-R2S%20RK3328%20OpenWrt%2019.07%20Build.yml#L61

删除部分package避免冲突
https://github.com/oimken/NanoPi-R2S/blob/4c48929e461b28b0aba21784855e3e23bca38794/.github/workflows/NanoPi-R2S%20RK3328%20OpenWrt%2019.07%20Build.yml#L115-L119

最后一段的错误：
```
020-10-06T07:39:37.4465643Z        ^~~~~~~~~~~
2020-10-06T07:39:37.4465895Z        LUA_ERRMEM
2020-10-06T07:39:37.4901680Z At top level:
2020-10-06T07:39:37.4903392Z src/hlua.c:4446:19: warning: 'hlua_applet_http_send_response' defined but not used [-Wunused-function]
2020-10-06T07:39:37.4904283Z  __LJMP static int hlua_applet_http_send_response(lua_State *L)
2020-10-06T07:39:37.4904790Z                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2020-10-06T07:39:37.4926024Z cc1: warning: unrecognized command line option '-Wno-cast-function-type'
2020-10-06T07:39:37.4944657Z make[4]: *** [src/hlua.o] Error 1
2020-10-06T07:39:37.4946255Z Makefile:886: recipe for target 'src/hlua.o' failed
2020-10-06T07:39:37.4947510Z make[4]: Leaving directory '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/haproxy-ssl/haproxy-2.1.7'
2020-10-06T07:39:37.4949929Z make[3]: *** [/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/haproxy-ssl/haproxy-2.1.7/.built] Error 2
2020-10-06T07:39:37.4951611Z Makefile:161: recipe for target '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/haproxy-ssl/haproxy-2.1.7/.built' failed
2020-10-06T07:39:37.4953057Z make[3]: Leaving directory '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/feeds/packages/net/haproxy'
2020-10-06T07:39:37.4955366Z time: package/feeds/packages/haproxy/ssl/compile#1.06#0.15#1.21
2020-10-06T07:39:37.4956455Z make[2]: *** [package/feeds/packages/haproxy/compile] Error 2
2020-10-06T07:39:37.4958889Z package/Makefile:111: recipe for target 'package/feeds/packages/haproxy/compile' failed
2020-10-06T07:39:37.4965458Z make[2]: Leaving directory '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt'
2020-10-06T07:39:37.4986699Z make[1]: *** [/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2020-10-06T07:39:37.4988382Z package/Makefile:107: recipe for target '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
2020-10-06T07:39:37.4994127Z make[1]: Leaving directory '/home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt'
2020-10-06T07:39:37.5015758Z make: *** [world] Error 2
2020-10-06T07:39:37.5017186Z /home/runner/work/NanoPi-R2S/NanoPi-R2S/code/friendlywrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-10-06T07:39:37.5028688Z Command exited with non-zero status 1
2020-10-06T07:39:37.5029141Z you take 2:46:07 to build friendlywrt
2020-10-06T07:39:37.5120918Z ##[error]Process completed with exit code 1.
2020-10-06T07:39:37.5442543Z Post job cleanup.
2020-10-06T07:39:38.0257637Z [command]/usr/bin/git version
2020-10-06T07:39:38.0318795Z git version 2.28.0
2020-10-06T07:39:38.0374254Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-10-06T07:39:38.0429735Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-10-06T07:39:38.0759413Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-10-06T07:39:38.0800341Z http.https://github.com/.extraheader
2020-10-06T07:39:38.0809218Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-10-06T07:39:38.0854139Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-10-06T07:39:38.1209246Z Cleaning up orphan processes
```

soffchen, 2020-10-06:
这个问题大概是 gcc 版本低了，回头我看看

tefiszx, 2020-10-07:
> @tefiszx 还是失败了
>
> 修改 `NanoPi-R2S RK3328 OpenWrt 19.07 Build.yml` [...]

这个错误是haproxy编译出错的提示，你所编译的进程还没编译到passwall这个阶段。你在config中去除haproxy的相关选项，再看看结果如何。

oimken, 2020-10-07:
@tefiszx   我明白，但其实我没有更改其他的设置，而是passwall依赖haproxy，取消haproxy也不会成功编译passwall.

oimken, 2020-10-07:
@soffchen  非常感谢！对于编译这事确实我是门外汉，您有空看看就最好不过了，主要是觉得passwall比较易用，用习惯了。

slb116763814, 2020-12-15:
PASSWALL这个我也想要的功能

soffchen, 2021-01-23:
新版集成了

closed 2021-01-24 by soffchen