soffchen/NanoPi-R2S #61: Action script is not working anymore

wikimao, 2020-12-29:
Action script is not working anymore, compile issue. Many dependencies are missing. Anyone encountered this problem?

soffchen, 2020-12-29:
I will try it at 2021.1.1

wikimao, 2020-12-29:
What’s your wechat ?

wikimao, 2021-01-04:
Hi, Did you try?

trytrojan, 2021-01-06:
install clash fail:

log:

Run cd code/friendlywrt/package/base-files/files
  cd code/friendlywrt/package/base-files/files
  mkdir -p etc/openclash/core
  wget -qO- https://github.com/Dreamacro/clash/releases/download/v1.2.0/clash-linux-armv8-v1.2.0.gz | gunzip -c > etc/openclash/core/clash
  wget -qO- https://github.com/vernesong/OpenClash/releases/download/TUN-Premium/clash-linux-armv8-2020.10.29.g5e54f48.gz | gunzip -c > etc/openclash/core/clash_tun
  wget -qO- https://github.com/vernesong/OpenClash/releases/download/TUN/clash-linux-armv8.tar.gz | tar xOvz > etc/openclash/core/clash_game
  chmod +x etc/openclash/core/clash*
  shell: /bin/bash -e {0}

gzip: stdin: unexpected end of file
Error: Process completed with exit code 1.

wangxin6, 2021-01-06:
> install clash fail:
>
> log:
>
> Run cd code/friendlywrt/package/base-files/files
> cd code/friendlywrt/package/base-files/files
> mkdir -p etc/openclash/core
> wget -qO- [...]

https://github.com/soffchen/NanoPi-R2S/issues/54

wikimao, 2021-01-07:
> #54
=====================================================
Update YML file with this: ( URL )

https://github.com/Dreamacro/clash/releases/download/v1.3.5/clash-linux-armv8-v1.3.5.gz
https://github.com/vernesong/OpenClash/releases/download/TUN-Premium/clash-linux-armv8-2021.01.01.g0ab75c5.gz
https://github.com/vernesong/OpenClash/releases/download/TUN/clash-linux-armv8.tar.gz

trytrojan, 2021-01-07:
> https://github.com/Dreamacro/clash/releases/download/v1.3.5/clash-linux-armv8-v1.3.5.gz
> https://github.com/vernesong/OpenClash/releases/download/TUN-Premium/clash-linux-armv8-2021.01.01.g0ab75c5.gz [...]

thanks!

wikimao, 2021-01-11:
> I will try it on 2021.1.1

hi, Did you try?

soffchen, 2021-01-24:
Updated

closed 2021-01-24 by soffchen