soffchen/NanoPi-R2S #72: How to update?

xenstar, 2021-02-08:
1) Is there any easy way to update it? Or I have to download the zip file, then burn it in the memory card, then restore my settings in openwrt. 

Please let me know.

soffchen, 2021-02-08:
There is an OTA tool called rs2 flasher in the System menu, please check it.

soffchen, 2021-02-08:
You may also upload the img.zip file, and use `rom_flash` command to update it.

xenstar, 2021-02-08:
> You may also upload the img.zip file, and use `rom_flash` command to update it.

This one I don't know, not good at coding.

xenstar, 2021-02-08:
> There is an OTA tool called rs2 flasher in the System menu, please check it.

Yes I found that r2s flasher, so I just upload the zip file via r2s and it will automatically update right? Will my settings be gone?

soffchen, 2021-02-08:
> > There is an OTA tool called rs2 flasher in the System menu, please check it.
> 
> Yes I found that r2s flasher, so I just upload the zip file via r2s and it will automatically update right? Will my settings be gone?

There is an option `Keep settings` checked by default.

xenstar, 2021-02-08:
> > > There is an OTA tool called rs2 flasher in the System menu, please check it.
> > 
> > 
> > Yes I found that r2s flasher, so I just upload the zip file via r2s and it will automatically update right? Will my settings be gone?
> 
> There is an option `Keep settings` checked by default.

great. let me try that. Thanks for quick reply. Do you have any telegram group?

xenstar, 2021-02-08:
[image]

Is there any way to know if it's updated or not? because after the update, the page didn't reload.

closed 2021-02-09 by soffchen