soffchen/NanoPi-R2S #73: docker 不能启动成功

zhixiangjoy, 2021-02-08:
大佬你好，r2s刷了你的固件，docker 还是不能正常启动，报错如下：
```
root@FriendlyWrt:~# dockerd
WARN[2021-02-08T09:12:45.005508162Z] failed to load plugin io.containerd.snapshotter.v1.btrfs  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2021-02-08T09:12:45.143900273Z] failed to load plugin io.containerd.snapshotter.v1.aufs  error="modprobe aufs failed: "": exit status 255"
WARN[2021-02-08T09:12:45.145710487Z] could not use snapshotter zfs in metadata plugin  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter: skip plugin"
WARN[2021-02-08T09:12:45.145831538Z] could not use snapshotter btrfs in metadata plugin  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2021-02-08T09:12:45.145882875Z] could not use snapshotter aufs in metadata plugin  error="modprobe aufs failed: "": exit status 255"
failed to start daemon: rename /opt/docker/runtimes /opt/docker/runtimes-old: invalid cross-device link
```
另外问下有交流群吗？

soffchen, 2021-02-08:
这个应该不是刷的最新版吧？

老版本可以试试 ```mount /dev/mmcblk0p2 /opt && mkdir /opt/docker && /etc/init.d/dockerd restart```

zhixiangjoy, 2021-02-08:
我记得下载的是2.7 最新版本的

zhixiangjoy, 2021-02-08:
> 这个应该不是刷的最新版吧？
> 
> 老版本可以试试 `mount /dev/mmcblk0p2 /opt && mkdir /opt/docker && /etc/init.d/dockerd restart`

按照你的方法 重新挂载卷后 docker正常启动了。

soffchen, 2021-02-08:
那就行，新版是会在第一次启动的时候执行这个初始化动作。

soffchen, 2021-02-08:
> 我记得下载的是2.7 最新版本的

应该是保留设置做的升级吧？那样不会执行初始化脚本。

zhixiangjoy, 2021-02-08:
> > 我记得下载的是2.7 最新版本的
> 
> 应该是保留设置做的升级吧？那样不会执行初始化脚本。

不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。

zhixiangjoy, 2021-02-08:
> 这个应该不是刷的最新版吧？
> 
> 老版本可以试试 `mount /dev/mmcblk0p2 /opt && mkdir /opt/docker && /etc/init.d/dockerd restart`

我重新下载最新固件，测试没有问题哈。可能是我下载了旧版固件。

zhixiangjoy, 2021-02-08:
另外请教一下怎么添加无线驱动进去？ 我有一块 comfast cf-811ac 的usb 无线网卡，目前能驱动2.4G。

soffchen, 2021-02-08:
> 另外请教一下怎么添加无线驱动进去？ 我有一块 comfast cf-811ac 的usb 无线网卡，目前能驱动2.4G。

没有什么好办法，官方是单独编译出来 ko，然后打包进去的，建议再买一块在支持列表的网卡吧 😁

soffchen, 2021-02-08:
> > > 我记得下载的是2.7 最新版本的
> > 
> > 
> > 应该是保留设置做的升级吧？那样不会执行初始化脚本。
> 
> 不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。
> 我一会儿重新再刷一次，看下能不能复现。

Actions 开始编译了，等编好了你去试试吧，没问题说一下，我发到 release 去 😁

soffchen, 2021-02-08:
> > > 我记得下载的是2.7 最新版本的
> > 
> > 
> > 应该是保留设置做的升级吧？那样不会执行初始化脚本。
> 
> 不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。
> 我一会儿重新再刷一次，看下能不能复现。

编译好 ，试试看？

zhixiangjoy, 2021-02-08:
> > > > 我记得下载的是2.7 最新版本的
> > > 
> > > 
> > > 应该是保留设置做的升级吧？那样不会执行初始化脚本。
> > 
> > 
> > 不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。
> > 我一会儿重新再刷一次，看下能不能复现。
> 
> 编译好 ，试试看？

完美，我通过系统-->r2s刷机刷入成功，dockerman和docker正常启动：

[image]
[image]

zhixiangjoy, 2021-02-08:
发现的问题：

- 貌似解锁wy歌曲 golang 版本不能正常启动，一直未运行

soffchen, 2021-02-08:
> 发现的问题：
> 
> * 貌似解锁wy歌曲 golang 版本不能正常启动，一直提示： UnblockNeteaseMusic 未运行

Actions 最新版应该可以了

soffchen, 2021-02-08:
> > > > > 我记得下载的是2.7 最新版本的
> > > > 
> > > > 
> > > > 应该是保留设置做的升级吧？那样不会执行初始化脚本。
> > > 
> > > 
> > > 不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。
> > > 我一会儿重新再刷一次，看下能不能复现。
> > 
> > 
> > 编译好 ，试试看？
> 
> 完美，我通过系统-->r2s刷机刷入成功，dockerman和docker正常启动：
> 
> [image]
> [image]
> 不过升级完成后,提示正在重启。系统是关机了，不是自动重启，我断电重新上电启动的，不知道是不是正常现象？

我也遇到过，得拔电重启，不知道为啥。

zhixiangjoy, 2021-02-08:
> > > > > > 我记得下载的是2.7 最新版本的
> > > > > 
> > > > > 
> > > > > 应该是保留设置做的升级吧？那样不会执行初始化脚本。
> > > > 
> > > > 
> > > > 不是，我是重新烧录镜像到sd卡上的，另外建议把：luc-app-dockerman 集成进去呢。
> > > > 我一会儿重新再刷一次，看下能不能复现。
> > > 
> > > 
> > > 编译好 ，试试看？
> > 
> > 
> > 完美，我通过系统-->r2s刷机刷入成功，dockerman和docker正常启动：
> > [image]
> > [image]
> > 不过升级完成后,提示正在重启。系统是关机了，不是自动重启，我断电重新上电启动的，不知道是不是正常现象？
> 
> 我也遇到过，得拔电重启，不知道为啥。

刚测试解锁歌曲正常了哈。

closed 2021-02-09 by soffchen